MarekKowalski/LiveScan3D-Hololens #3: compile errors in unity 5.6

heikog, 2017-04-05:
Hey,
i get multiple compile time errors on unity 5.6

can you give me a hint of what to do ?
[image]

MarekKowalski, 2017-04-05:
Hi, I tried compiling on Unity 5.6 and it works on my side (although there was an issue with a shader).
Could you try with the version that is currently in the repo? Also which Visual Studio are you using?

heikog, 2017-04-05:
thanks for quick response. compiles and deploys like a charm now - unity 5.6 VSW community 2017.
however i cant get it to show the pointclouds.
i started the client, connected to the server, started the live view on the server, then started the app in the hololense and put in the servers ip adress - but then nothing happens. any further idea where i could look ?

MarekKowalski, 2017-04-05:
I'm glad to hear it compiles now!
Some things you could look into:
- try recompiling the NetworkCommunication dll and make sure it gets built to /Assets/Plugin/x86/,
- check your firewall settings and the capabilities in Unity publishing settings,
- if none of the above works I suggest you try run LiveScanServer with a debugger and see if it receives the connection from the HoloLens,
- if you do not have VS2015 to compile LiveScanServer as in the previous step, you can try to debug the HololensReceiver while it is running on HoloLens and see if it ever connects.

Let me know which one was it, if any \.

heikog, 2017-04-05:
damn - never mind. it was behind me...

MarekKowalski, 2017-04-05:
Glad to hear you have found it! :)

closed 2017-04-05 by MarekKowalski