MarekKowalski/LiveScan3D-Hololens #6: Upgrade to Visual Studio 2015 or 2017

Andy1210, 2017-06-22:
Could you please update the project to support the new visual studio?

MarekKowalski, 2017-06-24:
Hi, both the NetworkCommunication and the Unity projects use Visual Studio 2015. Which project opens in a different Visual Studio for you?

Marek

Andy1210, 2017-11-13:
The HololensReceiver project. Can it be because I have community edition of VS? (message: This project is incompatible with the current edition of Visual Studio)

MarekKowalski, 2017-11-14:
Hi,

Which version of VS do you have? Is it 2015 or 2017 community, is it fully updated?
I have VS 2015 community and it works for me.

Marek

Andy1210, 2017-11-15:
I have both, and fully updated yes.

MarekKowalski, 2017-11-16:
Hi,

I have just realised that you wrote that you cannot open the HololensReceiver project, all the time I thought you mean the other one. That project is not meant to be opened with Visual Studio, you should open this entire directory with the Unity Editor.

Marek