MarekKowalski/LiveScan3D-Hololens #7: Programs exits after IP address

ptltorres, 2017-06-28:
Hi, 

I'm using my PC as a server. My Kinect and HoloLens are connected to the PC. The project works fine on Unity but when it asked me for the IP address the following appeared on the output window: " has exited with code -1073740791 (0xc0000409)" I thought I entered the wrong IP addres but the same happens with every different address. 

Thanks.

MarekKowalski, 2017-06-28:
Hi, 

Have you tried running the HoloLens application debugging mode to see where the crash happens?
Is the HoloLens on the same network as your computer?

Marek

ptltorres, 2017-06-29:
Never mind. It works perfectly. The issue was my school's networks that was blocking the communication between the devices. I tried at at home it everything went good. 

Thank you!

closed 2017-06-29 by MarekKowalski