MarekKowalski/LiveScan3D-Hololens #8: Application exits when runs on hololens but works perfectly in Unity Editor

krishnabhaskar, 2017-07-24:
Hi,
When application is deployed to Microsoft Hololens, app opens asking to enter IP address. When server IP address is entered, it exits the application within few seconds. But in unity editor point cloud is rendered perfectly.
Unity Version - 5.6.0
Visual Studio Version - Visual Studio Community 2015.
LiveScan3d SDK - Livescan3d-1.0.0-vc12
Build Type - XAML UWP
Could you please have a look into this issue.
Thanks

MarekKowalski, 2017-07-25:
Hi! Can you access the HoloLens device protal from the server machine? Also, have you checked the firewall settings?

Thanks,
Marek

krishnabhaskar, 2017-07-26:
Now it works. Thanks a lot !!!

krishnabhaskar, 2017-07-26:
Hi,
Can this sdk work with Kinect v1 ???
If no, can u please tell how any possibility with the source code to make it work with Kinect v1.

Thanks
Krishna

closed 2017-07-26 by krishnabhaskar

NYhale, 2024-07-02:
> 现在它起作用了。多谢！！！

I also encountered this problem, how did you solve it? thank you