MarekKowalski/LiveScan3D-Hololens #11: Build for OpenVR

neopostmodern, 2017-08-31:
We're trying to run this on VR HMDs (Vive, Rift...). Which works fine in the Unity Editor.
But, building hasn't worked out at all.

First, I ripped out all the Hololens related objects/scripts in the Unity project and instead called `Connect` manually (in the `PointCloudReceiver.cs`).  
Building for Windows (x64) failed because Unity won't include the referenced `platform.winmd` which appears to be a hard requirement of the NetworkCommunication project.  
Building via XAML and for PC only and messing a bit with the `csproj` (to rip out Hololens references) worked, but I won't see anything in the app, nor will it launch on the VR HMD.

Do you have any hints where to dig around in the project? Can you estimate how hard it would be to rip out the `platform.winmd` dependent code and re-write that?

MarekKowalski, 2017-09-06:
Hi!
I don't think you need the NetworkCommunication project if you are not running this as a Windows Universal Platform (windows store) application. I can't test it now since I am away till the end of September but I believe that if you build it as a standalone PC application it does not even use the NetworkCommuniaction project. The same should be the case for a VR build.

NetworkCommunication is there because I had trouble getting the UWP C# sockets to work efficiently on the HoloLens. If you run this on a PC, which is also the same PC you have the server application on, the standard sockets (the ones that run in the Editor mode) should be just fine.

Let me know if this makes sense to you.

Marek

neopostmodern, 2017-09-07:
Works like a charm! Simply excluded the NetworkCommunication libraries and build / running on VR-HMD worked immediately.
Thanks a lot for the hint — and the software in general!

closed 2017-09-07 by neopostmodern