MarekKowalski/LiveScan3D-Hololens #12: Update to Unity 2017, VS 2017, Latest Mixed Reality Toolkit

reillydonovan, 2017-09-06:
Hi, this is great!  The only problem is... I can't seem to get it to work on the hololens :(  
It runs well in the Unity editor, and I can build a visual studio solution, but when I try to deploy it to the hololens it ends in failure.  I have a feeling it must have something to do with a combination of conflicts regarding updates.  I'm using Unity 2017, VS 2017, and the latest mixed reality toolkit (however I am not importing this into your sample project).  Any suggestions on how to get this to deploy to the hololens?  Thanks!

reillydonovan, 2017-09-07:
This is what is returned shortly after the build begins, then crashes:

Severity	Code	Description	Project	File	Line	Suppression State
Error		The command ""C:\Users\reillydonovan\Downloads\LiveScan3D-Hololens-master\LiveScan3D-Hololens-master\HololensReceiver\Apps\Receiver01\Unity\Tools\AssemblyConverter.exe" -platform=uap -lock="C:\Users\reillydonovan\Downloads\LiveScan3D-Hololens-master\LiveScan3D-Hololens-master\HololensReceiver\Apps\Receiver01\HololensReceiver\project.lock.json" -bits=32 -configuration=Release -removeDebuggableAttribute=False -uwpsdk=10.0.15063.0 -path="." -path="C:\Program Files\Unity\Editor\Data\PlaybackEngines\MetroSupport\Players\UAP\dotnet\x86\Release" "C:\Users\reillydonovan\Downloads\LiveScan3D-Hololens-master\LiveScan3D-Hololens-master\HololensReceiver\Apps\Receiver01\HololensReceiver\Assembly-CSharp.dll" "C:\Users\reillydonovan\Downloads\LiveScan3D-Hololens-master\LiveScan3D-Hololens-master\HololensReceiver\Apps\Receiver01\HololensReceiver\UnityEngine.dll" "C:\Users\reillydonovan\Downloads\LiveScan3D-Hololens-master\LiveScan3D-Hololens-master\HololensReceiver\Apps\Receiver01\HololensReceiver\UnityEngine.Timeline.dll" "C:\Users\reillydonovan\Downloads\LiveScan3D-Hololens-master\LiveScan3D-Hololens-master\HololensReceiver\Apps\Receiver01\HololensReceiver\nunit.framework.dll" "C:\Users\reillydonovan\Downloads\LiveScan3D-Hololens-master\LiveScan3D-Hololens-master\HololensReceiver\Apps\Receiver01\HololensReceiver\UnityEngine.UI.dll" "C:\Users\reillydonovan\Downloads\LiveScan3D-Hololens-master\LiveScan3D-Hololens-master\HololensReceiver\Apps\Receiver01\HololensReceiver\UnityEngine.HoloLens.dll" "C:\Users\reillydonovan\Downloads\LiveScan3D-Hololens-master\LiveScan3D-Hololens-master\HololensReceiver\Apps\Receiver01\HololensReceiver\UnityEngine.Networking.dll"" exited with code 1.	HololensReceiver	C:\Users\reillydonovan\Downloads\LiveScan3D-Hololens-master\LiveScan3D-Hololens-master\HololensReceiver\Apps\Receiver01\HololensReceiver\HololensReceiver.csproj	279

MarekKowalski, 2017-09-08:
Hi,

I updated Unity, and it all builds fine for me with Visual Studio 2017 community.
I did not actually check if it works on the HoloLens as I do not have one with me now. I understand that for you it crashes in the building step? 

I would suggest trying to rebuild the NetworkCommunication library (you have to place the dll and winmd files in Assets/plugins/).
Also, did you remember to set XAML instead of D3D in Unity build settings? 

Let me know if I can help you further.

Marek

reillydonovan, 2017-09-08:
The error returned above is from Visual Studio when trying to deploy to the hololens.
I can successfully build a visual studio solution from Unity but I cannot successfully deploy the Visual Studio solution to the hololens.  

Are you using the latest windows 10 creators update?

I did change the settings to XAML instead of D3D.  What Windows SDK are you using?
Are there any additional settings for publishing that you are including?
I've been checking what Windows recommends for publishing to the holocene:

Virtual Reality Supported: Windows Holographic: 16 bit.

When you go to deploy the Visual Studio Solution do you change all of the recommended settings?  Windows.Universal to Windows.Holographic?  Debug to Release? ARM to x86?

To rebuild Network Communication should I just open it is VS and build it then copy and paste the dll as you describe?

Thanks a lot for your help, I can't wait to get this working!

reillydonovan, 2017-09-08:
So after rebuilding the Network Communication VS project and dropping in the .dll and winmd this is what I'm returned from VS when trying to build to the hololens:

Severity	Code	Description	Project	File	Line	Suppression State
Error		There was a mismatch between the processor architecture of the project being built "x86" and the processor architecture, "AMD64", of the implementation file "D:\UnityProjects\LiveScan3D-Hololens-master\HololensReceiver\Apps\Receiver08\HololensReceiver\NetworkCommunication.dll" for "D:\UnityProjects\LiveScan3D-Hololens-master\HololensReceiver\Apps\Receiver08\HololensReceiver\NetworkCommunication.winmd". This mismatch may cause runtime failures. Please consider changing the targeted processor architecture of your project through the Configuration Manager so as to align the processor architectures between your project and implementation file, or choose a winmd file with an implementation file that has a processor architecture which matches the targeted processor architecture of your project.	HololensReceiver

MarekKowalski, 2017-09-11:
Hi,

I built using SDK version 10.0.15063. I also change ARM to x86, it builds in both Debug and Release modes. 
The error you showed in the second message suggests that you have x64 set somewhere in the NetworkCommunication project. I can't tell where it is exactly, I suggest you try googling this error.

Let me know if there is any other way I can help.

Marek

reillydonovan, 2017-09-17:
The issue is the latest version of Visual Studio, version 1.11.33288.831
Using the previous version everything publishes perfectly, thanks for this great toolkit~!

MarekKowalski, 2017-09-17:
I am glad to hear you were able to solve it!

closed 2017-09-17 by MarekKowalski