MarekKowalski/LiveScan3D-Hololens #14: hololens can't get cloud point

iamsile, 2018-03-22:
Hi,@MarekKowalski, I use deploy the app on the hololens and input the server ip, but the hololens couldn't get any cloud points, and the server can run. I don't know where is wrong, Could you help me?

Thanks!

MarekKowalski, 2018-03-22:
Hi,

When you open the app on HoloLens a keyboard should appear and allow you to input the IP address of the server. In the recent versions of Unity the keyboard is unfortunately broken. For now you will have to remove the keyboard and specify the IP address in code and leave it there "hardcoded".

We will try to figure out a new solution soon.

Marek

iamsile, 2018-03-22:
@MarekKowalski  Yes, I had write the ip address into the code. but it doesn't work. I don't know how to do

MarekKowalski, 2018-03-22:
Hi,

I see, can you try placing a breakpoint in the TransferServer class of the LiveScanServer to see if it receives the connection from the HoloLens? Also make sure your firewall settings allow the connection.

Marek

iamsile, 2018-03-22:
@MarekKowalski  When I read the readme in the project. I found this sentence like "Next start the Unity application on the same machine as the server, if everything is fine it should be working."
Shoud I open a unity app to server like livescan3d server ? I don't understand this step.
Does open a unity app to compile project ?

MarekKowalski, 2018-03-22:
What this sentence means, is that you should start the Unity project in the editor (by pressing play over the scene).

iamsile, 2018-03-22:
"Next start the Unity application on the same machine as the server, if everything is fine it should be working."

Does use the Unity application as a server ?

iamsile, 2018-03-22:
@MarekKowalski  as I know, open the unity application only can compile the project.

iamsile, 2018-03-22:
Hi, @MarekKowalski 
My unity version is 5.5.2f and visual studio version is 2015
When I wrote the ip address into the code, I haven't get any crash and the hololens app always could run. But I can't see any cloud piont.

What should I do?

Thanks

MarekKowalski, 2018-03-22:
Hi, 

I have answered this question above. Please try to write only a single message with a question, instead of multiple ones. Each message you write gets sent to all of the people who watch this thread, it is annoying to get many.

Marek

iamsile, 2018-03-22:
Hi, @MarekKowalski  

1. I don't understand "Next start the Unity application on the same machine as the server, if everything is fine it should be working."
Open the Unity application on the same machine as the server,  does open it like LiveScan3d server ?

2. I had write the ip address into the code. built it and installed it. But when I open app on the hololens, I din't see any cloud point and the app runs well. I don't know how to find where it is wrong.

That's all my question. 

Look forward to your reply.

Thanks

MarekKowalski, 2018-03-22:
Hi,

I answered both of those questions above, **please read the replies more carefully**. I am posting the answers here again:
1. What this sentence means, is that you should start the Unity project in the editor (by pressing play over the scene).
2. I see, can you try placing a breakpoint in the TransferServer class of the LiveScanServer to see if it receives the connection from the HoloLens? Also make sure your firewall settings allow the connection.

Marek

iamsile, 2018-03-22:
Hi, @MarekKowalski  Thanks for your reply.

I had find why is wrong in my project. I can get the cloud point at now.

1. First wrong is that I don't close the firewall.
2. I don't comment the "TouchScreenKeyboard.Open" in the KeyboardInput.cs

Thank you very much, Thank you

closed 2018-03-22 by iamsile