MarekKowalski/LiveScan3D-Hololens #15: Building for iOS: "Shader Is Not Supported On This GPU (none of subshaders/fallbacks are suitable)"

nickcastel50, 2018-07-23:
Hello,

First, very cool project! I wanted to try building this for iOS, but I'm running into an issue. What I've done:

1. Removed NetworkCommunication Code (as referenced in another issue if we are not building for Windows Store)
2. Stripped Hololens Gesture Control code
3. Hardcoded IP - manually called connect function.

The project compiles fine, but I see a warning under the GSG Billboard shader object, saying: "Shader Is Not Supported On This GPU (none of subshaders/fallbacks are suitable)". Does anyone know the reason for this? I'm not quite sure why the iOS GPU wouldn't be able to properly render this shader. 

Thanks for any information you can provide, and thanks again for the cool project!

nickcastel50, 2018-07-26:
Thank you @MarekKowalski! I will try this and report back. I think I still have an error to fix, I was getting a black screen on iOS after the Unity logo appears and the logs say "SocketException:Connection Refused". This is strange to me because on the PC where I am running the server I see that the port connection is established and looking at Wireshark logs I can see the point cloud stream is being sent to my phone's IP over port 48002.

I thought even with the bad shader it should still render, albeit just a solid pink color? Maybe I'm wrong, will let you know as soon as I try with the shader you provided. Thanks again!

nickcastel50, 2018-07-28:
Okay @MarekKowalski, I got it rendering on the iPhone! Since I removed the gesture control code, I had to rotate the camera Y axis 180 degrees to see it. That shader is still an issue though, it just renders depth, it doesn't render color at all. 

Also one other thing, the background is black. Would you know would I go about making the background transparent and having the background be my phone's camera feed for some sort of AR experience? Thanks again for your help and project!

TheBricktop, 2018-08-14:
How did You manage to autoconnect at start, it always freezes my editor when I try to get it like this

MarekKowalski, 2018-08-18:
@nickcastel50, congratulations on your progress and sorry I did not answer, for some reason I missed your post completely :( Unfortunately I know little a bout shaders as well, so I won't be able to help you there.

@TheBricktop , have you made sure the live view window is open and showing the point cloud (on the same device as Unity)? If you did and it does not work, please check your firewall settings.

Marek

nickcastel50, 2018-11-10:
> How did You manage to autoconnect at start, it always freezes my editor when I try to get it like this

@TheBricktop: You can just just modify the KeyboardInput.cs file in @MarekKowalski's project. Just call keyboardDone.invoke(YOUR_IP_ADDRESS); in the start() function. It works fine for testing. 

Something like this:

```
void Start ()
    {
        if (keyboardDone != null)
            keyboardDone.Invoke(YOUR_IP_ADDRESS);
    }
	
void Update ()
    {
     
    }
```