MarekKowalski/LiveScan3D-Hololens #16: SocketException:Unable to connect because the target computer is actively rejected

StoneClose, 2018-09-25:
System.Net.Sockets .Socket.Connect (System.Net .EndPoint remoteEP, Boolean requireSock etPolicy)
System.Net.Sockets .Socket.Connect (System.Net.EdPoin remoteEP)
System.Net.Sockets TcpClient.Connect (System .Net.IPEndPoint remote_ end_ point)
System.Net .Sockets .TcpClient.Connect System.Net.IPAddres] ipAddresses, Int32 port)
I tried many ways, but I couldn't solve the problem,can you help me?Thanks

MarekKowalski, 2018-09-26:
Hi,

When is this happening? Are you trying to connect from the HoloLens to the PC or is it from the PC (in Unity) to the same PC? Did you check the firewall settings?

Marek

tkbala, 2019-02-01:
Hi,

I am facing the same issue. I have two systems - One is a laptop that is connected to the kinect, and the other is a PC that is running this program. I am trying to get this to run in Unity (No Hololens yet). I tried with both Unity 5.6 as well as Unity 2018.3. The firewall is turned off. Any ideas?

Thanks,
Bala

tkbala, 2019-02-01:
It looks like @StoneClose  might not be opening the server from the same PC as the unity application (or) the server might not have been started.