MarekKowalski/LiveScan3D-Hololens #20: Will you support Hololens2?

flufy3d, 2019-08-14:


MarekKowalski, 2019-08-14:
I'm not sure any changes would be necessary for HoloLens2, other then switching to the ARM architecture.

PinoEire, 2020-07-28:
Network Communication fails to build in ARM64, any pointers?

MarekKowalski, 2020-08-10:
Hi,

Sorry for taking a while to answer, I just tried compiling NetworkCommunication for ARM64 and it worked after a few changes.
The things I changed were:
 - Target platform version - I changed to the latest one, but some older ones would likely work as well.
 - Platform toolset - changed to Visual Studio 2019.
Here is a picture with the two fields marked in red:
[image]

CopperfieldZ, 2020-12-02:
Hello，when will you release an version for HL2?I am looking forward to it!

MarekKowalski, 2020-12-13:
Hi,
Unfortunately I do not have access to the hardware due to continued remote work.
Once I get the access I hope to update to HL2. 
Having said that, I feel the update should be quite easy. If you do have the hardware, you might want to give it a shot. I'd be very happy to help.

Marek

EzioZz, 2021-01-13:
I'm working on deploying this project on hololen2.
Here is my questions:
1. Which is the root path fo the project, that I open with unity："..\LiveScan3D-Hololens" or  "..\LiveScan3D-Hololens\HololensReceiver"
2. what should I do with,  "NetworkCommunication". Should I compile it?
[image]

EzioZz, 2021-01-14:
I have deployed the APP on the Hololens2. But it will crash once I opened it.
May be it is caused by the architecture of the unity plugin name "NetworkCommunication". It should be Arm on Hololens2.
[image]

MarekKowalski, 2021-01-22:
Hi,

Yes, NetworkCommunication should definitely be built for Arm to work on HL2.
Let me know if that works and if not what error you are seeing.

Thanks,

Marek

closed 2021-01-29 by flufy3d

fberna, 2021-03-18:
Hi Marek,

any news about the release for HL2?
Hope everything is fine.

FB

MarekKowalski, 2021-03-29:
I updated [this ](https://github.com/MarekKowalski/LiveScan3D-Hololens/issues/26) issue with information about the release.