MarekKowalski/LiveScan3D-Hololens #21: Building and Deploying the LiveScan3D-Hololens (Windows 10 and Unity)

sarkrishnan, 2019-11-29:
@MarekKowalski I have tried to open the LiveScan3D-Hololens with different versions of Unity.

Unity 5.6.1f1 crashes when I open the the project. Same with Unity 5.6.0 too with same result. 

Can you please say which windows version you used originally or the combined version of windows and unity (5.6.0 or Unity 5.6.1f1,...) that worked for you? I tried in Windows 10 Enterprise 64-bit (10.0, Build 17763).

Thank you and cheers.

MarekKowalski, 2019-12-02:
Hmm, that's a weird one. There is no error from Unity, right? Does it crash immediatelly when you open the directory? Have you tried removing some of the files and then opening? What compilation errors do you get?

I'm not sure which Unity version Iw as using for this in the past. 

Marek

MarekKowalski, 2019-12-02:
Looks like someone got the project to open in 5.6.0f3 recently: https://github.com/MarekKowalski/LiveScan3D-Hololens/issues/22

sarkrishnan, 2019-12-05:
Thanks @MarekKowalski I have reached out to the issue opener in #22

sarkrishnan, 2020-02-19:
Hi @MarekKowalski 

I am able to successfully deploy the app in Windows 10. I am giving the details of the version I used for everyone's reference.

Unity - 2018.4.14f1
Visual Studio - Microsoft Visual Studio Community 2019 Version 16.4.2 

Cheers,
Aravinda 

P.S: I am closing this issue since I was able to build and deploy in Windows 10

closed 2020-02-19 by sarkrishnan