MarekKowalski/LiveScan3D-Hololens #22: Issues building with Unity 5.6.0f3 & Visual Studio 2017

tinuxnet, 2019-12-02:
1>  Copying unprocessed assemblies...
1>  Running AssemblyConverter...
1>  System.Collections.Generic.KeyNotFoundException: The given key was not present in the dictionary.
1>     at System.Collections.Generic.Dictionary`2.get_Item(TKey key)
1>     at UnityEditor.Scripting.Compilers.NuGetPackageResolver.Resolve()
1>     at Unity.UWPAssemblyResolver..ctor(String projectLockFile, String desiredUWPSDKVersion)
1>     at Unity.OperationContext.SetPlatform(Platform platform, String projectLockFile, String uwpSDK)
1>     at Unity.ParseArgsStep.Execute()
1>     at Unity.Step.Execute(OperationContext operationContext, IStepContext previousStepContext)
1>     at Unity.Operation.Execute()
1>     at Unity.Program.Main(String[] args)

error MSB3073

How to solve it? Any tips?
In the meantime, is there somewhere a pre-build package in order to test it on HoloLens?

Thank you

MarekKowalski, 2019-12-02:
Hi, unfortunately there is no pre-built package, that's a pretty good idea though.
One thing I would try is rebuilding the NetworkCommunication.dll using VS 2017. 
Please let me know if I can help more or if you manage to solve this.

sarkrishnan, 2019-12-05:
@tixunet Can you please mention the unity version, visual studio version and windows version you utilized? I do not want to repeat myself. Please refer to the issue I opened #21  Thank you.

hanseuljun, 2020-08-28:
See here for the temporary fix for this issue: https://forum.unity.com/threads/net-scripting-backend-and-visual-studio-2017-3-incompatibility.487833/

This is an incompatibility issue between .Net UWP solution built by Unity 5.6 and newer versions of Visual Studio. I think updating the Unity version of this project to a version that Unity built while newer versions of Visual Studio were available would fix this, while I do not mean that would be a must as it might involve a lot of work...