MarekKowalski/LiveScan3D-Hololens #24: Having difficulty connecting HoloLens to Azure Kinect

hanseuljun, 2020-08-28:
Thanks for providing source code of this project. I was trying to build this repository to use it together with LiveScan3D. I built the AuzreKinect branch of LiveScan3D and it works well. The problem is when I try to build LiveScan3D-Hololens for a HoloLens. To make it clear, the project works when it is inside Unity and even when the visual studio version is running it is inside the a HoloLens emulator.

Inside HoloLens, it stops working here:
https://github.com/MarekKowalski/LiveScan3D-Hololens/blob/d93cb12d80567c6b1e065d1da005feffc3aaa05a/HololensReceiver/Assets/PointCloudReceiver.cs#L68
After typing the IP address and submitting it, I checked it goes to the line in front of the above with a Unity I added. I turned off all firewall and also checked the advanced settings for LiveScanServer (allowing both public and private incoming packets).

As it is code from NetworkCommunication.dll, I tried adding dependency dlls of NetworkCommunication (i.e., msvcp140_app.dll, vccorlib140_app.dll, vcruntime140_app.dll), but it was not a fix. For the build, I used Unity 2018.4.14., Visual Studio 2019, and Windows SDK 10.0.18362.0. I used the .NET scripting backend for UWP from Unity.

Would there be any suggestions?
I can think of getting inside NetworkCommunication or using IL2CPP scripting backend, but not sure they make sense...

hanseuljun, 2020-08-28:
I fixed this. Application crashes when endpoint that the machine cannot connect is given. This happens when PointCloudReceiver tries to connect -> tries to construct a TransferSocket -> consturctor of SocketClient inside socketCS.cpp throws an expection since due to failure of connection. I guess this is because socketCS was not built with Unity in mind, a C# framework were you cannot catch a native exception...

My fix was changing the port of LiveScan3D to one that the network I use allows.

closed 2020-08-28 by hanseuljun

MarekKowalski, 2020-09-07:
Thanks for explaining what the issue was!