MarekKowalski/LiveScan3D-Hololens #25: there's nothing in the hololens after inputing ip

Sunpro108, 2021-01-10:
I build the vs project successfully and I could see the panel to input the ip. But after that, there is noting at all.

MarekKowalski, 2021-01-11:
Hi,

As explined in the manual (in the docs folder), there are a few apps in the VS project.
You have to start at least one instance of the client app (used for grabbing Kienct output) and one instance of the server app (which combines outputs from all Kinects). The client app (the one with the IP address window) should show you the current output image from the Kinect, if it doesn't then you should see an error on the status bar.

Once both apps are started, you should press start on the server and the connect on the client.
Once that's done you can see the Kinect outputs in the live view window on the server.

Sunpro108, 2021-01-12:
I had got the 3d scene and it isexcellent! But I try to deploy the app to hololens , I could not see any thing in the hololens.

EzioZz, 2021-03-21:
Have you tried to turn off the firewall ?

Slayer016, 2022-02-07:
I am a Highschool student, trying to replicate this for a Science fair project, and have the same issue. I am new to programming!! My env: Unity 2018.4.14f1, VS 2019, Hololens 2. Have 2 Kinect cameras, can see the holograph in the Scene in Unity.
 I was able to build the package, and deploy to HL2, and open it. I am able to input the IP, and then in the App, but do not see anything. I have disabled the firewall, added the device IP as a trusted source in Norton 360, still cannot see anything.
Appreciate your help and guidance.
PS. I tried several times and failed with newer versions of Unity and OpenXR (did not know enough to change GestureManager.cs from VR to XR and all the deprecated methods, and hence went to the 2018.4.14f1 version. If you have a newer version of that file that I can use, I can try with HL2 and newer code.

Thanks