MarekKowalski/LiveScan3D-Hololens #29: How do I get stream to flow into Unity

myheathcliff, 2022-06-29:
Sorry, I have never learned Unity before. I have installed all the required applications (LiveScan3D /Kinects V2 / Unity-2018.4.14C1 /Visual Studio 2019), but when I opened Unity there was no response. Is there anything else I need to do after I open Unity? thank you

MarekKowalski, 2022-06-30:
I suggest you checkout a basic tutorial of Unity, it would make your life a little easier :)
Here's a list of steps you would need to take:
- Setup Livescan3d and open the live view window
- Open Unity and direct it at the Unity project that came with this repo
- Open the scene by selecting the scene file in Unity assets window (it will be on the bottom of the screen and show some files)
- Start the Unity app by pressing the play button over the main preview window.

myheathcliff, 2022-07-01:
Thank you very much for your instruction. I'll have a try！