MarekKowalski/LiveScan3D-Hololens #31: Point cloud can't not show in hololens2

Hans86, 2022-09-28:
I've tried to use the hololens2 branch in unity and it works,but when I built them in hololens2, there was no reaction.
Then I rebuilt the NetworkCommunication in Release mode,but it failed and I could only ues Debug mode to rebuild.
The release mode error is MSB3073.

chuan-chu, 2022-10-14:
What is your unity version? I am trying with 2020.3, seems not working any more. GestureSettings.NavigationX etc seems no longer supported?

MarekKowalski, 2022-10-27:
Unfortunately I have not tried running this in a while and I can't remember what I used last. Let me know if you figure it out and I'll update the readme.

simahsa, 2023-02-09:
I have the exact same problem (error MSB3073 while rebuilding the NetworkCommunication). Were you able to figure this out?

MarekKowalski, 2023-02-22:
Can you post the whole error from the build process?

simahsa, 2023-02-22:
Thank you so much for your time! I was able to deploy the app in HL2, but
after I enter the IP address, I cannot see anything. I see the discussion
in your GitHUB and it was recommended to build the network
communication; however, when I build it (Release/ARM64) I receive this
error message:

*MSB3073* The command "copy C:\Users\University of
NM\Documents\GitHub\LiveScan3D-Hololens\NetworkCommunication\\Release\NetworkCommunication.dll
C:\Users\University of
NM\Documents\GitHub\LiveScan3D-Hololens\NetworkCommunication\\..\HololensReceiver\Assets\Plugins\x86\
copy C:\Users\University of
NM\Documents\GitHub\LiveScan3D-Hololens\NetworkCommunication\\Release\NetworkCommunication.winmd
C:\Users\University of
NM\Documents\GitHub\LiveScan3D-Hololens\NetworkCommunication\\..\HololensReceiver\Assets\Plugins\x86\
:VCEnd" exited with code 1. NetworkCommunicationWinRT C:\Program Files
(x86)\Microsoft Visual
Studio\2019\Community\MSBuild\Microsoft\VC\v160\Microsoft.CppCommon.targets
155

Is there anything that I am missing? Thank you in advance.

Best regards,
Mahsa

MarekKowalski, 2023-02-22:
Looks like what fails is a post-build step that copies the DLL + winmd file into the unity project. Do the directories referenced in the copy command exist? Does "C:\Users\University of NM\Documents\GitHub\LiveScan3D-Hololens\NetworkCommunication\\Release\NetworkCommunication.dll" exist?
I suspect the issue may be due to spaces in the path as command line arguments do not like spaces.

simahsa, 2023-02-22:
Yes, it does. As I can deploy the app on Hololens2, maybe the problem is
not related to network communication. I also have some warning in Unity
saying that the 'Gesture manager' and 'object manipulator' are obsolete in
this version of unity (2018.4.14f1). I am guessing maybe the reason that I
cannot see anything on HL2 is because of that?

MarekKowalski, 2023-03-13:
Did you manage to resolve this in the end? Unfortunately I have not had the bandwidth to run this app on HL2 recently, so not sure I can help with the HL2 aspect of it.

simahsa, 2023-06-13:
> Did you manage to resolve this in the end? Unfortunately I have not had the bandwidth to run this app on HL2 recently, so not sure I can help with the HL2 aspect of it.

Yes, we did. The main issue was that it was using the previous version of mixed reality without ever updating it to MRTK2. Once we updated it to MRTK2 it worked perfectly.

GitMaxOch, 2024-04-15:
Hey Mahsa, I am also trying to convert the program to a newer version but i have some issues. How did you resolve the problem with the gesture control. thanks in advance.