MarekKowalski/LiveScan3D-Hololens #34: Unity problem DirectoryNotFoundException: Could not find a part of the path

MrBred, 2023-11-25:
Hi,

Thank you for the really impressive application, I was very happy when I found it. I am an enthusiastic layman but I managed to use the LiveScan application, but I got stuck with Unity.

I have a Hololens 2 so I would like to be able to connect it but I couldn’t.

The LiveScan application is running, I see the live image, I can save ply, which I can play.

However, I ran into a problem with the LiveScan3D-Hololens application. The LiveScanServer is running, I see the live image. I open the project folder in Unity but I get stuck there. I get 6 error messages but each one starts the same way.

“DirectoryNotFoundException: Could not find a part of the path”

Unfortunately, based on this, I’m afraid I won’t be able to connect the thing with the Hololens either. Could you help me with what to install on the Hololens and exactly how to run the programs?
[image]

Thank you for your answer,
Benedek

MarekKowalski, 2023-12-07:
Hello!

Thanks trying the app and asking this question. Few questions:
1. Are you on the hololens 2 branch?
2. Did you try re-build the NetworkCommunication project?

Thanks,

Marek

MarekKowalski, 2023-12-07:
Are you using the hololens 2 branch of this repo? The NetworkCommunication project, once built, will produce a DLL file that goes into Assets/Plugins (it should be copied there automatically, feel free to check). This DLL will enable communication between the app on HL2 and the LiveScan app running on your PC.

What happens when you start the app on HL2? If you run in Debug mode do you see a point at which there's an exception or something like that?

MrBred, 2023-12-08:
I used wrong repo... But, unfortunately, the new HL2 repo doesn’t work either (I donwloaded Unity 2019.4.11f1) This is the error message in Unity (from Visual Studio): Debug.LogError($“For Mixed Reality Toolkit input to work properly, please remove your other input module(s) and add a {typeof(MixedRealityInputModule).Name} to your main camera.”, inputModules[0]); 

Also, I can’t run the Release on NetworkCommunication. The attached picture is the error.

[image]

What is the right sequence? (when open the projekt with unity, when relase the network communication, when attach to unity the holo reciver, have to make build and i have to debug on hololens? I can install the wrong repo to HL but this don't work...)

Is it works with LiveScan player or just server?

Sorry for my unundersantding.

MarekKowalski, 2023-12-18:
As for this error:
Debug.LogError($“For Mixed Reality Toolkit input to work properly, please remove your other input module(s) and add a {typeof(MixedRealityInputModule).Name} to your main camera.”,
unfortuantely, I am not sure what it is, I'd suggest looking on stackoverflow ,etc.

As for running the networkcommunication project: it is not meant to be started directly. That project produces a dll that is consumed by Unity. The .DLL output by the project should be placed in the Assets/Plugins directory automatically, once you build NetworkCommunication. Does that happen? Is it build with ARM (as it should by for HL2)? If that all happens then you can close that project.

At this point you should:

- open Unity
- build the Unity project for UWP platform, this will create a Visual Studio project
- open the Visual Studio project, build it and play it on the hololens.

Should work with both LiveScan player and server.