MarekKowalski/LiveScan3D-Hololens #36: Very low fps

lex6i, 2024-05-27:
Hi, I'm trying to stream pointcloud data from two kinects 2 (on two computers) and the framerate is super low (1-3), any idea of how to improve that?  Both the computers are quite powerful

MarekKowalski, 2024-05-29:
Hey! Is this on HL2 or desktop? If on desktop, what is the framerate in the LiveScanServer app?  Is everything compiled in Releae?

lex6i, 2024-05-29:
It's on desktop,  that is the framerate in the livescanserver app.  It's about twice as fast if only taking bodies.  but still not usable result for me.  It's from the prebuilt binaries.

MarekKowalski, 2024-05-29:
Interesting, that should not be happeing. I wonder if it could be a question of network bandwidth maybe? 
How much network usage are you seeing? Also, how much CPU / GPU usage?